marekburiak/ILI9341_due #1: for DUE: appearently cursorToXY does not work

VogonJeltz, 2015-09-19:
hi,
for the DUE:
running the following code your ILI9341_due outputs only text the first time  on the first line.
then no text is displayed any more.
This code lets one define different display times 
by 
LCDTYPE = _ILI9341due_;
I am choosing your lib,
by
LCDTYPE = _ILI9341_;
I choose the standard Adafruit lib which works perfectly.

Anything is going wrong with your's:

~~~
#include <SPI.h>
#include <SD.h>
#include <ardustdio.h>

//=====================================================================================
// misc.  
//=====================================================================================

#define  _DUEMISO_    74  // Arduino Due SPI Header
#define  _DUEMOSI_    75
#define  _DUESCK_     76

#define  _MEGAMISO_   50  // Arduino MegaSPI Header
#define  _MEGAMOSI_   51
#define  _MEGASCK_    52

#define  clock()      millis()  
#define  LRAND_MAX    32767
#define  srand(seed)  randomSeed(seed)
#define  rand()       random(LRAND_MAX)
#define  rando()      ((float)rand()/(LRAND_MAX+1))

//=====================================================================================
// TFT LCD
//=====================================================================================

#define   UTFT_SmallFont     8 // UTFT 8x10
#define   UTFT_MediumFont   12 // UTFT ++ 
#define   UTFT_BigFont      18 // UTFT +++ 
#define   _SmallFont_        1 // 9341 6x9
#define   _MediumFont_       2 // 9341 12x16
#define   _BigFont_          3 // 9341 18x23

int16_t  LCDmaxX , LCDmaxY ;                // display size
int16_t  _curx_, _cury_,                    // last x,y cursor pos on TFT screen
         _maxx_, _maxy_;                    // max. x,y cursor pos on TFT screen       
char     wspace[128];                       // line of white space

// set LCD TFT type
int16_t  LCDTYPE   =   -1;

#define  _LCD1602_    1  // LCD1602  Hitachi HD44780 driver <LiquidCrystal.h> 
                           // http://www.arduino.cc/en/Tutorial/LiquidCrystal   //
#define  _SERLCD_     2  // Sparkfun serLCD 16x2  
                           // http://playground.arduino.cc/Code/SerLCD   //
#define  _UTFT_       4  // Henning Karlsen UTFT 2.2-2.4" 220x176 - 320x240 lib
                           // http://henningkarlsen.com/electronics/library.php?id=51   //
#define _ILI9341_     8  // https://github.com/adafruit/Adafruit_ILI9340
                           // https://github.com/adafruit/Adafruit-GFX-Library //
#define _ILI9341due_  9  // ILI9341_due NEW lib by Marek Buriak
                           // http://marekburiak.github.io/ILI9341_due/ //
                           
//--------------------------------------------------------------------------------------------------
#define    tft_cs     49
#define    tft_dc     48
[... 369 lines cut ...]
   
   sprintf(sbuf, "setup(): done.");
   Serial.println(); Serial.println(sbuf);   
   curlf(); curlf(); lcdprint(sbuf);

}
//=====================================================================================

//=====================================================================================

//=====================================================================================

//=====================================================================================
void loop(){
   char     sbuf[128];
  
   

}
//=====================================================================================
//=====================================================================================

~~~

If

marekburiak, 2015-09-19:
I have removed all other code that does not relate to ILI9341_due and I got the result you can see on the attached screenshot. Were you expecting something else? 
Also couple of notes.You can change the orientation by casting 0,1,2,3 to iliRotation enum:dtft.setRotation((iliRotation)orient);

There are tons of colors defined in ILI9341_due.h so you can use:ILI9341_BLACKILI9341_BLUE
ILI9341_CHOCOLATE

Here's my modified code (change DC, CS and RST):

#include <SPI.h>
// ILI9341_due NEW lib by Marek Buriak http://marekburiak.github.io/ILI9341_due/#include "ILI9341_due_config.h"#include "ILI9341_due.h"#include "SystemFont5x7.h"//#include "Streaming.h"// For the Adafruit shield, these are the default.

// UTFT Demo ported to ILI9341_due library by TFTLCDCyg
// Based on Demo 320x240 Serial of UTFT library // UTFT-web: http://www.henningkarlsen.com/electronics
//=====================================================================================// misc.  //=====================================================================================
#define  _DUEMISO_    74  // Arduino Due SPI Header#define  _DUEMOSI_    75#define  _DUESCK_     76
#define  _MEGAMISO_   50  // Arduino MegaSPI Header#define  _MEGAMOSI_   51#define  _MEGASCK_    52
#define  clock()      millis()  #define  LRAND_MAX    32767#define  srand(seed)  randomSeed(seed)#define  rand()       random(LRAND_MAX)#define  rando()      ((float)rand()/(LRAND_MAX+1))

//=====================================================================================// TFT LCD//=====================================================================================
#define   UTFT_SmallFont     8 // UTFT 8x10#define   UTFT_MediumFont   12 // UTFT ++ #define   UTFT_BigFont      18 // UTFT +++ #define   _SmallFont_        1 // 9341 6x9#define   _MediumFont_       2 // 9341 12x16#define   _BigFont_          3 // 9341 18x23
int16_t  LCDmaxX, LCDmaxY;                // display sizeint16_t  _curx_, _cury_,                    // last x,y cursor pos on TFT screen_maxx_, _maxy_;                    // max. x,y cursor pos on TFT screen       char     wspace[128];                       // line of white space

 // set LCD TFT typeint16_t  LCDTYPE = -1;
#define  _LCD1602_    1  // LCD1602  Hitachi HD44780 driver <LiquidCrystal.h> // http://www.arduino.cc/en/Tutorial/LiquidCrystal   //#define  _SERLCD_     2  // Sparkfun serLCD 16x2  // http://playground.arduino.cc/Code/SerLCD   //#define  _UTFT_       4  // Henning Karlsen UTFT 2.2-2.4" 220x176 - 320x240 lib// http://henningkarlsen.com/electronics/library.php?id=51   //#define _ILI9341_     8  // https://github.com/adafruit/Adafruit_ILI9340// https://github.com/adafruit/Adafruit-GFX-Library //#define _ILI9341due_  9  // ILI9341_due NEW lib by Marek Buriak// http://marekburiak.github.io/ILI9341_due/ //
//--------------------------------------------------------------------------------------------------#define    tft_cs     10#define    tft_dc     9#define    tft_rst     8                        
//=====================================================================================// UTFT Henning Karlsen//=====================================================================================//#include <UTFTQD.h>//////UTFT   qdUTFT(Model,  SDA=MOSI,   SCL,      CS,     RESET,  RS)    // Due: 3 exposed SS pins: 4,10,52////UTFT   qdUTFT(QD220A,   A2,       A1,       A5,     A4,     A3);   // adjust model parameter and pins!//UTFT   qdUTFT(QD220A, _MEGAMOSI_, _MEGASCK_, tft_cs, 0, 49);   // A0->Vc (LED), A4->BoardReset//extern  uint8_t SmallFont[];//////=====================================================================================//// TFT Adafruit LIL9340/ILI9341////=====================================================================================//#include <Adafruit_GFX.h>//#include <Adafruit_ILI9340.h>////Adafruit_ILI9340  tft = Adafruit_ILI9340(tft_cs, tft_dc, tft_rst);

ILI9341_due      dtft = ILI9341_due(tft_cs, tft_dc, tft_rst);
// Color set#define  BLACK           0x0000#define RED             0xF800#define GREEN           0x07E0//#define BLUE            0x001F#define BLUE            0x102E#define CYAN            0x07FF#define MAGENTA         0xF81F#define YELLOW          0xFFE0 #define ORANGE          0xFD20#define GREENYELLOW     0xAFE5 #define DARKGREEN       0x03E0#define WHITE           0xFFFF
uint16_t  color;
//--------------------------------------------------------------------------------------------------
#define  lcdWhiteBlack()  { if(LCDTYPE==_ILI9341due_)   { dtft.setTextColor(WHITE, BLACK) ;}}
#define  lcdNormal()      { if(LCDTYPE==_ILI9341due_)   { dtft.setTextColor(WHITE, BLACK) ;}}
#define  lcdInvers()      {if(LCDTYPE==_ILI9341due_)   { dtft.setTextColor(BLACK, WHITE) ;}}
#define  lcdWhiteRed()    {if(LCDTYPE==_ILI9341due_)   { dtft.setTextColor(WHITE, RED) ;}}
#define  lcdRedBlack()    {if (LCDTYPE == _ILI9341due_) { dtft.setTextColor(RED, BLACK); }}
#define  lcdYellowBlue()  {if (LCDTYPE == _ILI9341due_) { dtft.setTextColor(YELLOW, BLUE); }}
int16_t  fontwi = 8;  // defaultint16_t  fonthi = 10;  // default

void putfonttype(uint8_t fsize) { if (LCDTYPE == _ILI9341due_) { if (fsize == _SmallFont_) { fontwi = 6; fonthi = 9; }  // 5x7 + overhead  } _maxx_ = LCDmaxX / fontwi;    // max number of letters x>> _maxy_ = LCDmaxY / fonthi;    // max number of letters y^^  memset(wspace, ' ', _maxx_);  // line of white space wspace[_maxx_] = '\0';}

void setlcdorient(int16_t orient) {
[... cut ...]
                           // http://www.arduino.cc/en/Tutorial/LiquidCrystal   //
                           // http://playground.arduino.cc/Code/SerLCD   //
                           // http://henningkarlsen.com/electronics/library.php?id=51   //
                           // https://github.com/adafruit/Adafruit-GFX-Library //
                           // http://marekburiak.github.io/ILI9341_due/ //

//--------------------------------------------------------------------------------------------------

VogonJeltz, 2015-09-19:
could you pls put it in code tags 
~~~

~~~
I'll try it ASAP !

marekburiak, 2015-09-19:
I have removed all other code that does not relate to ILI9341_due and I got the result you can see on the attached screenshot. Were you expecting something else? 
Also couple of notes.You can change the orientation by casting 0,1,2,3 to iliRotation enum:dtft.setRotation((iliRotation)orient);

There are tons of colors defined in ILI9341_due.h so you can use:ILI9341_BLACKILI9341_BLUE
ILI9341_CHOCOLATE

Here's my modified code (change DC, CS and RST):

~~~
#include <SPI.h>

// ILI9341_due NEW lib by Marek Buriak http://marekburiak.github.io/ILI9341_due/
#include "ILI9341_due_config.h"
#include "ILI9341_due.h"
#include "SystemFont5x7.h"
//#include "Streaming.h"
// For the Adafruit shield, these are the default.

// UTFT Demo ported to ILI9341_due library by TFTLCDCyg

// Based on Demo 320x240 Serial of UTFT library 
// UTFT-web: http://www.henningkarlsen.com/electronics

//=====================================================================================
// misc.  
//=====================================================================================

#define  _DUEMISO_    74  // Arduino Due SPI Header
#define  _DUEMOSI_    75
#define  _DUESCK_     76

#define  _MEGAMISO_   50  // Arduino MegaSPI Header
#define  _MEGAMOSI_   51
#define  _MEGASCK_    52

#define  clock()      millis()  
#define  LRAND_MAX    32767
#define  srand(seed)  randomSeed(seed)
#define  rand()       random(LRAND_MAX)
#define  rando()      ((float)rand()/(LRAND_MAX+1))

//=====================================================================================
// TFT LCD
//=====================================================================================

#define   UTFT_SmallFont     8 // UTFT 8x10
#define   UTFT_MediumFont   12 // UTFT ++ 
#define   UTFT_BigFont      18 // UTFT +++ 
#define   _SmallFont_        1 // 9341 6x9
#define   _MediumFont_       2 // 9341 12x16
#define   _BigFont_          3 // 9341 18x23

int16_t  LCDmaxX, LCDmaxY;                // display size
int16_t  _curx_, _cury_,                    // last x,y cursor pos on TFT screen
_maxx_, _maxy_;                    // max. x,y cursor pos on TFT screen       
char     wspace[128];                       // line of white space

											// set LCD TFT type
int16_t  LCDTYPE = -1;

#define  _LCD1602_    1  // LCD1602  Hitachi HD44780 driver <LiquidCrystal.h> 
// http://www.arduino.cc/en/Tutorial/LiquidCrystal   //
#define  _SERLCD_     2  // Sparkfun serLCD 16x2  
// http://playground.arduino.cc/Code/SerLCD   //
#define  _UTFT_       4  // Henning Karlsen UTFT 2.2-2.4" 220x176 - 320x240 lib
// http://henningkarlsen.com/electronics/library.php?id=51   //
#define _ILI9341_     8  // https://github.com/adafruit/Adafruit_ILI9340
// https://github.com/adafruit/Adafruit-GFX-Library //
#define _ILI9341due_  9  // ILI9341_due NEW lib by Marek Buriak
// http://marekburiak.github.io/ILI9341_due/ //
[... 200 lines cut ...]
	
	sprintf(sbuf, "SD OK ! ");
	Serial.println(sbuf);
	curlf(); lcdprint(sbuf);

	sprintf(sbuf, "setup(): done.");
	Serial.println(); Serial.println(sbuf);
	curlf(); curlf(); lcdprint(sbuf);

}
//=====================================================================================

//=====================================================================================

//=====================================================================================

//=====================================================================================
void loop() {
	//char     sbuf[128];

}
//=====================================================================================
//=====================================================================================

~~~

VogonJeltz, 2015-09-19:
that looks really excellent!
thank you very much for your efforts!
I hope you don't assume my reports to be nitpicking, I was just reporting issues I was running into and I'd wish them to be considered to be constructive.
I'll try them out ASAP tomorrow. Thanks again!

VogonJeltz, 2015-09-20:
results:

your code works fine on my Due, also printAt(string, x, y) works fine!
Thank you very much, issue is reolved!

closed 2015-09-20 by marekburiak