marekburiak/ILI9341_due #2: function names

VogonJeltz, 2015-09-19:
I would suggest to use the same function names as how they are used in commen libs, e.g. Adafruit ILI9340.
https://github.com/adafruit/Adafruit_ILI9340
https://github.com/adafruit/Adafruit-GFX-Library

Instead it's weird if one always has to rewrite all the existing code when switching the display libs.

marekburiak, 2015-09-19:
I am sorry but I won't be changing function names at this point. Many people use the library and I don't want them to rewrite their code. Besides, I don't think many people switch between display libraries.

   

https://github.com/adafruit/Adafruit_ILI9340

VogonJeltz, 2015-09-19:
maybe have them as an alias additionally?

marekburiak, 2015-09-19:
I think it would just confuse people, have 2 functions with the same functionality. You can always modify my library on your machine and add any aliases you want in ILI9341_due.h somewhere at the end so you can easily copy it if you need to.

VogonJeltz, 2015-09-19:
no, thanks, that's too much effort if there will be updates - both for you and for me.
IMO it's always good to have possibly the  same syntax like those which are already established.
I saw that you once even changed some calls. e.g. setCursor to cursorToXY. Having both versions (and other ones ) for a limited time parallelly would be a good choice though, and putting the other one into a "will-extinct" lib for a while .
It's really hard for new users to apply to your quite unfamiliar API syntax while your current users surely quickly apply to the well-known Adafruit names.

marekburiak, 2015-09-19:
The cursorToXY in particular was taken from GLCD library which I have uses as a base for rendering the text. I don't think that my api is "quite unfamiliar", most of the graphic functions like drawCircle use the same names as in adafruit libraries. The text is rendered in a completely different way and I have text functions that adafruit does not have so I can't have 100% same naming scheme.Btw. instead of cursorToXY(..) and print(....) you can just use printAt(...).

VogonJeltz, 2015-09-19:
ok, thanks, and of course I respect that it's your lib ba your efforts and by your rules :)

closed 2015-09-19 by marekburiak