marekburiak/ILI9341_due #4: SD card issue

VogonJeltz, 2015-09-20:
I encountered the reason why my further test programs didn't show correct display outputs:
it happened when a SD card was plugged to the SPI bus additionally.

this is the SD init function:
~~~

int16_t initSD() {
   char sbuf[128];
   uint32_t  tstamp;
   int16_t   ior=0;
  
   tstamp = clock();
   ior=SD.begin(sd_cs);  // true on sucess; else false
   while( !ior) {      
      sprintf(sbuf, "#: ...SD not found... ");  
      curlf(); lcdprint("#: ...SD not found... "); 
      Serial.println(sbuf);
      delay(1000);   
      ior=SD.begin(sd_cs);
      if (clock()-tstamp>20000) {Serial.println("#: ...break!"); break; }
   } 
  if(!ior) return fileIO_ERR_SDCARD;     // SD ioresult==0 => error = -16 
  return fileIO_OK ;                     // SD ioresult==1 => ok = 1
} 

~~~
this is the call in setup():

~~~
// SD card
   sprintf(sbuf, "SD init... ");   Serial.println(sbuf);
   i = initSD();
   if(i==fileIO_ERR_SDCARD) sprintf(sbuf, "SD failed! ERROR ! ");  
   else sprintf(sbuf, "SD OK ! ");   
   Serial.println(sbuf);   
   curlf(); 
   lcdprint(sbuf);
~~~

it works fine with Adafruit ILI9340, but with ILI9341_due the program blocks then.

Instead if I outcomment the SD init line
// i = initSD();

then again the display works also fine with ILI9341_due.

What do I have to change to make the SPI bus work with both the ILI9341_due and the SD ?

marekburiak, 2015-09-20:
Try to uncomment this line in ILI9341_due_config.h:

~~~
//#define ILI_USE_SPI_TRANSACTION
~~~

If that won't work then SD library must be overriding something that ILI9341_due uses. I had no problems using SDFat library together with my lib (sdFatTftBitmap example uses it to render images from SD card): https://github.com/greiman/SdFat
Btw. SdFat can use DMA on Due to read/write faster from/to SD card. That's where I got my initial DMA code for my lib from.

VogonJeltz, 2015-09-20:
is already uncommente by default!
// uncomment if you want to use SPI transactions. Uncomment it if the library does not work when used with other libraries.
//#define ILI_USE_SPI_TRANSACTION

marekburiak, 2015-09-20:
By default it's like this (commented out):
~~~
//#define ILI_USE_SPI_TRANSACTION
~~~

So try to uncomment it:

~~~
#define ILI_USE_SPI_TRANSACTION
~~~

VogonJeltz, 2015-09-20:
actually I don't want to change the SD lib because I need a lot of SD.h features for my program and I invested lot of time to have stdio.h compatible commands to access it, like
long     fprintf_( File * stream, const char fmtstr[], ... );       // see ANSI C: fprintf()
long     fscanf_ ( File * stream, const char fmtstr[], ... ) ;      // see ANSI C: fscanf()
char   * fgets_  ( char * str, int32_t num, File * stream );        // see ANSI C: fgets()
File     fopen_  ( char * filename, const char * mode );            // see ANSI C: fopen()
int16_t  fclose_ ( File   file_);                                   // see ANSI C: fclose()   
int16_t  remove_ ( char * filename );                               // see ANSI C: remove()

http://www.mindstormsforum.de/viewtopic.php?f=78&t=8491#p66429

VogonJeltz, 2015-09-20:
aah, yes, I will do !

VogonJeltz, 2015-09-20:
if uncommented: SD seems to work but then display text blocks instead.

marekburiak, 2015-09-20:
I am sure SDFat has those or similar functions too. If you don't want to switch SD library then I guess you'll have to find out why it's not working ;) Try to change the order in which you initialize tft and sd. If I remember you are initializing tft first and sd after that so try to initialize sd first and then tft.

VogonJeltz, 2015-09-20:
I had a look at this SDfat lib but that's really weird to me - no chance.
SD is fine, I will have to keep it.

I changed the init order, that seems to work better at the first glance.
I'll check it further on and will report!

VogonJeltz, 2015-09-20:
I had a look at this SDfat lib but that's really weird to me - no chance.
standard SD.h is fine, I will have to keep it.

update:

no, faulty, too: no display output any more after initianlizing the SD card.

VogonJeltz, 2015-09-20:
this is the complete test code, you can choose display type either
~~~
_ILI9341_ // Adafruit
// or
_ILI9341due_ // your lib
~~~

~~~
#include <SPI.h>
#include <SD.h>
//#include <ardustdio.h>

//=====================================================================================
// misc.  
//=====================================================================================

#define  _DUEMISO_    74  // Arduino Due SPI Header
#define  _DUEMOSI_    75
#define  _DUESCK_     76

#define  _MEGAMISO_   50  // Arduino MegaSPI Header
#define  _MEGAMOSI_   51
#define  _MEGASCK_    52

#define  clock()      millis()  
#define  LRAND_MAX    32767
#define  srand(seed)  randomSeed(seed)
#define  rand()       random(LRAND_MAX)
#define  rando()      ((float)rand()/(LRAND_MAX+1))

//=====================================================================================
// TFT LCD
//=====================================================================================

#define   UTFT_SmallFont     8 // UTFT 8x10
#define   UTFT_MediumFont   12 // UTFT ++ 
#define   UTFT_BigFont      18 // UTFT +++ 
#define   _SmallFont_        1 // 9341 6x9
#define   _MediumFont_       2 // 9341 12x16
#define   _BigFont_          3 // 9341 18x23

int16_t  LCDmaxX , LCDmaxY ;                // display size
int16_t  _curx_, _cury_,                    // last x,y cursor pos on TFT screen
         _maxx_, _maxy_;                    // max. x,y cursor pos on TFT screen       

// set LCD TFT type
int16_t  LCDTYPE   =   -1;

#define  _LCD1602_    1  // LCD1602  Hitachi HD44780 driver <LiquidCrystal.h> 
                           // http://www.arduino.cc/en/Tutorial/LiquidCrystal   //
#define  _SERLCD_     2  // Sparkfun serLCD 16x2  
                           // http://playground.arduino.cc/Code/SerLCD   //
#define  _UTFT_       4  // Henning Karlsen UTFT 2.2-2.4" 220x176 - 320x240 lib
                           // http://henningkarlsen.com/electronics/library.php?id=51   //
#define _ILI9341_     8  // https://github.com/adafruit/Adafruit_ILI9340
                           // https://github.com/adafruit/Adafruit-GFX-Library //
#define _ILI9341due_  9  // ILI9341_due NEW lib by Marek Buriak
                           // http://marekburiak.github.io/ILI9341_due/ //
                           
//--------------------------------------------------------------------------------------------------
#define    tft_cs     49
#define    tft_dc     48
#define    tft_rst     0                        
                          
//=====================================================================================
// UTFT Henning Karlsen
//=====================================================================================
#include <UTFTQD.h>

//UTFT   qdUTFT(Model,  SDA=MOSI,   SCL,      CS,     RESET,  RS)    // Due: 3 exposed SS pins: 4,10,52
[... 329 lines cut ...]
   sprintf(sbuf, "LCD=%d wi%dxhi%d Font %dx%d",LCDTYPE,LCDmaxX,LCDmaxY,fontwi,fonthi);
   Serial.println(sbuf); 
   Serial.println();
   lcdcls(); lcdprint(sbuf);  // LCD runstate
   
   curlf(); lcdprint(sbuf2);  // SD runstate

   
   sprintf(sbuf, "setup(): done.");
   Serial.println(); Serial.println(sbuf);   
   curlf(); curlf(); lcdprint(sbuf);

}

//=====================================================================================
//=====================================================================================
void loop(){
   char     sbuf[128];
  
   

}
//=====================================================================================
//=====================================================================================
~~~

marekburiak, 2015-09-20:
Just FYI, I won't have time to look into SD library and why it conflicts with my lib so I suggest you either switch SD library to SDFat or try to figure out why those two libs don't work together. Sorry but I have really no time at the moment (or any time soon).

VogonJeltz, 2015-09-20:
ok, don't mind, but I will use the Adafruit lib then if I need to have SD card access.
Thanks for your reply!

VogonJeltz, 2015-09-24:
**the test program also blocks using SdFat with your ILI9341_due lib,**
but nevertheless, Adafruit ILI9340 and UTFT are also fine with SdFat instead!

So there must be issues in your ILI9341_due lib which conflits with SD card backpanes in general running on the same SPi bus simultaneously!

marekburiak, 2015-09-24:
I am using SDFat to load images from SD card and that happens in an interleaving fashion so read a few bytes from SD card, send it to TFT, read next a few bytes from SD, send it to TFT...and it works fine. Maybe you can compare how I initialize TFT and SD card in sdFatTftBitmap example because it is definitely not blocking there.

VogonJeltz, 2015-09-24:
I'm using simply the standard SD initializing function
~~~
int ior=SD.begin(sd_cs);  // true on sucess; else false
~~~
I just call this, and the SD always works fine then, with either lib, no issues about SD (neithe SD.h nor SdFat.h)
Also Serial output is fine.

Just the screen output is blocking by your ILI9341_due lib, and TFT values are shown faulty:
~~~
Serial.println(" done.");   lcdcls();
   sprintf(sbuf, "LCD=%d wi%dxhi%d Font %dx%d",LCDTYPE,LCDmaxX,LCDmaxY,fontwi,fonthi);
~~~
~~~
(...)
LCD=9 wi0xhi0 Font 6x9

SD init... 
SD OK ! 

setup(): done.

~~~

VogonJeltz, 2015-09-24:
perhaps the issue was a syntax error related to Adafruit syntax.
This different function syntax for each single lib is really confusing.

closed 2016-04-11 by marekburiak