marekburiak/ILI9341_due #5: high-level API functions for a touch button lib?

VogonJeltz, 2015-09-29:
hi,
a discusion in the forum lead me to this UTFT code:
~~~
void loop()
{
  int but1, but2, but3, but4, butX, butY, pressed_button;
  boolean default_colors = true;
 
  but1 = myButtons.addButton( 10,  20, 300,  30, "Button 1");
  but2 = myButtons.addButton( 10,  60, 300,  30, "Button 2");
  but3 = myButtons.addButton( 10, 100, 300,  30, "Button 3");
  but4 = myButtons.addButton( 10, 140, 300,  30, "Button 4", BUTTON_DISABLED);
  butX = myButtons.addButton(279, 199,  40,  40, "a", BUTTON_SYMBOL);
  butY = myButtons.addButton(  0, 199, 100,  40, "I", BUTTON_SYMBOL | BUTTON_SYMBOL_REP_3X);
  myButtons.drawButtons();

 
  while(1)
  {
    if (myTouch.dataAvailable() == true)
    {
      pressed_button = myButtons.checkButtons();

      if (pressed_button==butX)
      {
        if (myButtons.buttonEnabled(but4))
          myButtons.disableButton(but4, true);
        else
          myButtons.enableButton(but4, true);
      }
      else if (pressed_button==butY)
      {
        if (default_colors)
        {
          myButtons.setButtonColors(VGA_YELLOW, VGA_RED, VGA_YELLOW, VGA_BLUE, VGA_GRAY);
          myButtons.relabelButton(butY, "_");
          myButtons.drawButtons();
          default_colors=false;
        }
        else
        {
          myButtons.setButtonColors(VGA_WHITE, VGA_GRAY, VGA_WHITE, VGA_RED, VGA_BLUE);
          myButtons.relabelButton(butY, "I");
          myButtons.drawButtons();
          default_colors=true;
        }
      }
      if (pressed_button==but1)
        // do stuff for but1
      if (pressed_button==but2)
        // do stuff for but2
      if (pressed_button==but3)
        // do stuff for but3
      if (pressed_button==but4)
        // do stuff for but4
      if (pressed_button==-1)
        // do stuff for nothing pressed
    }
  }
}
~~~
moreover, he pointed out that you already included UTFT libs into yourr code .
So would it be possible zo habe those UTFT funtions with your ILI9341_due lib as well?

marekburiak, 2015-09-29:
I don't plan do add those function since these are higher level graphic objects. I think it would be more appropriate to create a new library for these (buttons, checkboxes, radiobuttons, dropdowns, etc.) rather than sticking all code in ILI9341_due lib. If all you need is just the buttons then you can look at the code for those in UTFT and replace all UTFT drawing functions with ILI9341_due ones (like drawLine, fillRect etc..).
UTFT.h was a leftover from the conversion and I have removed the reference.

VogonJeltz, 2015-09-30:
thank you, but that's a pity, because ILI9341due_Buttons would perfectly 
fit to your already exuisting libs, just like UTFT_Buttons fits to Henning Karlsen UTFT libs.
Again a pity too, that the API function calls are all different to all existing libbs ( ILI9341due, UTFT, Adafruit_ILI9341. 
An identical API syntax would be very much appreciated and would make those things a lot easier!
But Henning Karlsen's code is actually big enough to sink a ship, no chance for me as an unexperienced hobby  user to alter these libs.

marekburiak, 2015-09-30:
You only need to convert UTFT_Buttons.cpp (and .h) which is only 272 lines long. The api functions are similar, like drawBitmap in UTFT is basically drawImage in ILI9341_due, drawRoundRect, drawRect are in both, just the arguments are bit different, UTFT uses (x1,y1,x2,y2) and ILI9341_due uses (x,y,w,h), UTFT uses setColor whereas ILI9341_due is passing color as argument in each drawing call, etc.Good exercise to get to know API of both libraries better (which you have to learn anyway).

VogonJeltz, 2015-09-30:
OMG... ;)
BTW, x1,y1,x2,y2 is far better and more common than (x,y,w,h),
and setColor is far better and more common than passing color as argument in each drawing call

marekburiak, 2015-09-30:
I would not say x1,y1,x2,y2 is far better, quite the contrary. You usually end up doing (x1,y1,x1+w,y1+h). UTFT_buttons is an example of it. Also if I want to draw a 10px rectangle at coordinates (122,56) the arguments will look like this: (122, 56,10,10). With x1,y1,x2,y2 it has to look like this (122, 56, 131, 65). In my option this is not very intuitive piece of code for a 10px rectangle. And if you want to change position you have to update all 4 arguments. Of course there are instances where x1,y1,x2,y2 are beneficial but I found those instances rare. I was originally using x1,y1,x2,y2 in my lib as well but then I decided to switch to x,y,w,h which I find much better. Everybody has an opinion I guess. Also all graphic APIs that I worked with (like DirectX, OpenGL, Swing, HTML5, Win32, .NET,...) use (x,y,w,h) so I would not say it more common either. Maybe in that little Arduino TFT world...
setColor is in general probably better but it's too much work to redo the lib and docs now.

VogonJeltz, 2015-09-30:
ok, you're probably right, it's about "that little Arduino TFT world"...
but in general, it would be better if you's applied 100% to UTFT syntax or 100% to Adafruit ILI9341 libs, andd if you not created a 3rd or 4th cross-over variant additionally... 8-)

marekburiak, 2015-09-30:
Honestly, I do not really care how other libraries are designed. I did it to suit my needs and being 100% compatible with other libs was not one of my initial "requirements" since I haven't even thought of releasing the lib to public at that time. You actually got lucky that there is at least some compatibility (most of the graphic function names are the same) because my initial version of the lib was based on ILI9341_t3 library which uses the same function names as for example Adafruit_ILI9341. If I built it from scratch, there would be no compatibility at all :) I also won't implement things in my own lib which I think are not of the best design (like x1,y1,x2,y2) so the API will be different. Thanks for your opinion though :)

VogonJeltz, 2015-09-30:
thank you also for your reply and your opinion as it's finally your own effort - but anyway, this attitude is finally the basic issue about all the Ardiuino libs for all hardware that exists all over the world: 
"jeder kocht sein eigenes Süppchen", as we say, what means something like "each do their own thing", but more pointedly.

marekburiak, 2015-09-30:
I think most of the individuals that cook their own soup are doing it so it tastes the best for them and they cook it in a way they think it's the best. Compatibility with other soups is not so important unless one of your goals is a wide-spread adoption of your recipe for various reasons. That was certainly not my goal. The good thing about different soups is that you get different flavors and you learn different ways of cooking it. That will ultimately help you to cook your own soup with a flavor you like the best and in a way you think it's the best. The thing is that most of the soup eaters just want a restaurant.

VogonJeltz, 2015-09-30:
yes, about soups you're right, but I can handle each soup by identical tools - i.e., the spoon, either if I cooked it by my own or have a soup at a restaurant.

About TFTs the tools would be the programming tools, i.e., the libs, and there are myriads of different ones by syntax.
Compared to tools for soups, it was if one was to have different special spoons to be able to eat either soup - and that aint no good IMO (CMIIW) :P

marekburiak, 2015-09-30:
Well, I hope that my soup is not *that* different to other soups, you certainly should not need a fork to eat it. Unless I cooked it too thick for your taste :)

VogonJeltz, 2015-09-30:
no, surely not too thick, as finally it's amazingly "quick" (speaks more for a thin super-liquid soup);
just the button ib is a little too cumbersome and unwieldy with all wide code deposits (corrwect Google English?) - a spoon with a fork and a knife side-by-side would be actually needed for it, plus a affixed spaghetti-code-winder/coiler underneath, perhaps.

Maybe you'd wish to think again about a UTFT_Button-like ILI9341_due touchbutton lib, nevertheless ?
;-)

marekburiak, 2015-09-30:
I can think about it but I have no time for it :)

VogonJeltz, 2015-09-30:
think less, spare time for work :)

closed 2016-04-11 by marekburiak