marekburiak/ILI9341_due #6: _spiSettings uninitialized

Pablo2048, 2016-03-11:
Hi Marek. Thank You for Your work, but it seems that I found a problem. In SPI transaction mode there is some initialization of _spiSettings missing in begin(...) method.

marekburiak, 2016-03-12:
Are you getting an error or somethings not working?

Pablo2048, 2016-03-13:
I'm trying to port Your library to Arduino ESP8266 platform. My TFT is version with touch panel included, so beacuse small I/O pins count on ESP I have to share SPI between TFT and Touch, so I have to use transactions. I'm actually in testing phase so I found out that _spiSettings are not initialized in .begin() - after studying SPI.h from Arduino ESP8266 there is default setting (in object constructor) for _bitOrder LSBFIRST, but ILI9341 need MSBFIRST.

ppostuma, 2016-05-24:
Hi Marek - fantastic work! I too am interested in a ESP8266 version (my new toy of choice is the Digistump Oak, which I hope I'll be working with for quite some time, and using for quite some projects). I have gotten the basic Adafruit library to work but yours looks far more useable and powerful. I would be happy to send you one if you're interested in looking at it. Pablo, if you're reading, have you had any luck?

Pablo2048, 2016-05-25:
Hi ppostuma - well, sort of. I have partially working code, but is way slower than optimized Adafruit variant (AS8).

marekburiak, 2016-05-25:
I have a few ESP8266 myself and was thinking to port the library for it but unfortunately I have no time for it at the moment (other projects in the way).

ppostuma, 2016-05-26:
understand! thanks again