marekburiak/ILI9341_due #8: Printing slowdown

nacho-rz, 2016-07-07:
First, I would like to thank you for this amazing library but it seems that I have encountered a problem.

All of a sudden the printing speed slowed down substantially without modifying any code.
`	gTextArea tempTextArea{ 9, 209, 69, 229 };
	tft.setTextArea(9, 209, 69, 229);
	tft.clearTextArea();
	tft.setTextColor(ILI9341_SNOW);
	tft.print("2224");
	Serial.println("printed!");`

This is a sample code that can't even print fast enough before the next cycle begins.
Funny things is that it was working perfectly before.

marekburiak, 2016-07-07:
I haven't changed anything in my lib in a quite a few months so not sure what timeframe you mean by "before". Have you not changed some settings in ILI9341_due_config.h or maybe added another lib that might change SPI settings or changed your code in loop() or added interrupts?  Could be many things.

nacho-rz, 2016-07-07:
I don't believe I have done anything.

Can you give me a sample code of a rapidly changing text?

marekburiak, 2016-07-07:
sorry, I don't have a sample but your example is pretty near. You don't need to clearArea in each loop. Just print over the existing text. So do this in setup()
tft.setTextArea(9, 209, 69, 229); tft.clearTextArea(); tft.setTextColor(ILI9341_SNOW);
and this in loop:
tft.print("text");
Threre are other functions that allow you to clear the previous text if it was longer than the new text (look at gTextEraseFromBOL, gTextEraseToEOL and gTextEraseFullLine).

nacho-rz, 2016-07-07:
I figured it out! Thank you.

it seems that your library deletes the previously written string unlike the original Adafruit library.

The thing that you have created is a marvel !

closed 2016-07-07 by nacho-rz