marekburiak/ILI9341_due #9: ESP8266

stanzlavos, 2017-02-28:
Is it possible to get this working on a NodeMCU ESP8266 ? :)

marekburiak, 2017-02-28:
I am sure it is possible. The question is, who's got time to do it? :)

stanzlavos, 2017-02-28:
Let me rephrase :D

Does the ESP8266 support DMA? In their documentation I did see DMA support
but only in the context of I2S interface.

marekburiak, 2017-02-28:
Unfortunately I don't know.

hasenradball, 2018-04-11:
Hi together,
Is there anything new to this topic?
I would like to run the library on an ESP8266 and ESP32.
Does it actually work now?

marekburiak, 2018-04-12:
nah, don't have time for it at all. Maybe the generic  ILI9341_SPI_MODE_NORMAL works but I haven't tried.

hasenradball, 2018-04-12:
Hi, no problem. I can try it.
I could also support. Do you have an Idea which special points to look at?

hasenradball, 2018-04-12:
Sorry Support is the wrong word.
I mean I can have a look at if the ESP works with the library.
But do you know from your experience which points have considered at first if it does not work.

marekburiak, 2018-04-12:
I guess it depends on what compilation errors you get :)

fotosettore, 2018-11-25:
hi,
any news about development and test of this library in esp32 and/or esp8266 ?

marekburiak, 2018-11-25:
Hi, nope, nothing new. I have absolutely no time (and honestly also very small desire) to look at it.