marekburiak/ILI9341_due #19: transparent text mode glitch

ImpulseAdventure, 2019-07-29:
Hello Marek! -- Thank you for producing this well-optimized display library! The performance is excellent and it is good to see the thought that has gone into the justified text enhancements.

I am in the process of integrating support for **ILI9341_due** into the [GUIslice](https://github.com/ImpulseAdventure/GUIslice) GUI library but ran into what appears to be an issue with `gTextFontModeTransparent` mode rendering.

## Observation
- It appears that certain narrow characters in the `SystemFont5x7` are rendering with vertical line "glitches" on the right edge of the character extents when in transparent text mode.
- Solid text mode does not render these glitches.

[image]

## Debug
I haven't yet tried to debug the `drawTransparentChar()` routine in detail, but I suspect there is an issue that is triggering the following clause erroneously, with `lineEnd=6`:
[image]
My guess would be that a character with a bit set at the bottom row of the previous column followed by a column with no bits set causes lingering state that is detected in the full vertical strip clause above.

## Details
- **Library Version**: v1.01.009 https://github.com/marekburiak/ILI9341_due/commit/4a9d67fe1ed8d8061367be8076babc18299dfebe
- **MCU**: Arduino Due
- **TFT**: [Adafruit 2.8" Capacitive shield](https://www.adafruit.com/product/1947)

## Steps to Reproduce
- Run the attached sketch.
[due-font-glitch.zip](https://github.com/marekburiak/ILI9341_due/files/3440460/due-font-glitch.zip)

Thanks again very much for the library!

marekburiak, 2019-08-01:
Thanks for reporting the issue Calvin. Not sure when I'll be able to look at it though. If you get tired of waiting for the fix and decide and manage to fix it yourself then feel free to send me the fix and I'll update the library on GitHub with your name mentioned in the readme for eternity ;)
And just out of curiosity, does it only happen with SystemFont5x7 or with other (custom) fonts as well? I haven't used this font much while developing so that's probably why I haven't noticed it.

ImpulseAdventure, 2019-08-06:
Hi Marek! I think I have identified a suitable fix and am in the process of testing it with other fonts. If it continues to look stable, I will file a PR for your repo.

I believe the issue can theoretically occur in any fonts on characters that have a high-order bit set in the vertical raster, followed by a blank column. As the SystemFont5x7 is monospaced, there is a greater likelihood (versus proportional fonts) of having characters that encode trailing blank padding columns.

thanks!

ImpulseAdventure, 2019-08-12:
Closing the issue as Marek has merged my PR #20 . Thanks!

closed 2019-08-12 by ImpulseAdventure