marekburiak/ILI9341_due #23: Teensy support

Jonage, 2020-04-04:
This is utterly awesome library. I'm really missing it on Teensy. Is there any alternative?
Using basic Adafruit basic library is like living in a stone age.

jonnystokes, 2020-04-05:
How can this library be made compatible with teensy 4.0?

marekburiak, 2020-04-05:
Oh yea, it can be made compatible, the question is...who's gonna do it :)

Jonage, 2020-04-05:
I tried... But - it is toooooo much :-D

marekburiak, 2020-04-05:
I agree...somebody needs to spend considerable knowledge, time, patience and enthusiasm to get this converted to teensy. And I have none of these at this time :)

closed 2020-04-05 by marekburiak

ImpulseAdventure, 2020-04-05:
You are right... Marek’s library is definitely a fantastic library!

In the interim, if you need an optimization for Teensy, you may want to also take a look at ili9341_t3. Porting can be a lot of work (as evidenced by the PJRC threads), especially if aiming to retain the level of performance that Marek has already achieved for us with the Due and other AVRs. Have fun :)